MicroCoreLabs/Projects #5: ADC/SBC buggy?

GrantMeStrength, 2021-03-24:
Hi -

I am wondering if you have checked your (beautifully neat) 6502 implementation against a respected test such as [Klaus2m5](https://github.com/Klaus2m5/6502_65C02_functional_tests/blob/master/6502_functional_test.a65)?

I ask because I have been testing my own emulation, and I have trouble with the notorious ADC/SBC opcodes. When I tried your implementations it also failed. 

I see in your code the line:

 **if ((0x00FF&bcd_total) > 0x09) { bcd_total=bcd_total+0x010; bcd_total=bcd_total-0x0A; }**    

which contains a reference **bcd_total** before it has been assigned a value, thus the condition is never true. Maybe you missed something?

thanks!

-John

MicroCoreLabs, 2021-03-24:
Hi,

Thanks for your email!   Youre correct that the bcd_total is left-over code which is not used.  I will clean it up...

Yes,  I definitely ran Klause's tests and they all passed.. even the BCD as far as I know... I think I also ran another BCD test too...

I will check again though.. maybe I passed Klaus's test with the BCD tests optioned out... Ill take a look.

Were you able to get my code to pass the BCD test?  If so, what was the mistake in my code? 🙂

Thanks again,
-Ted

GrantMeStrength, 2021-03-24:
I am still trying to get ANY BCD code to pass the tests, mine or yours. If yours is good, the issue is elsewhere in my code and I apologize for suggesting otherwise :-)  

I know that the first time I ran Klause's test my code passed and I was delighted - and then I remembered BCD testing was off, and turned it on, and was less delighted ;-)

MicroCoreLabs, 2021-03-25:
Im confident that you'll get it. 🙂      Thanks for catching the stale code in my BCD algorithm!

Best of luck!

-Ted

closed 2021-11-21 by GrantMeStrength