MicroCoreLabs/Projects #13: M68K - Infinite loop in test all opcodes

xahon, 2023-02-09:
I have an endless loop in this part of code

[image]

lines 3898-3916
https://github.com/MicroCoreLabs/Projects/blob/ea66a81165a3f88b8d14fafd05a566c4d131c5ae/MCL68/MC68000_Test_Code/MC68000_test_all_opcodes.X68#L3898-L3916

Easy68k v5.16.1

MicroCoreLabs, 2023-02-10:
It seems like is running forever, but it is just running many iterations.  If you run in Easy68K you will see it gets past this code and is not an infinite loop.

xahon, 2023-02-13:
You are right. It takes a lot of patience to finish the program. I accidentally forgot to turn off the simulation for a night and now it's infinitely loops in ALL_DONE branch

closed 2023-02-13 by xahon