MicroCoreLabs/Projects #14: [MCL65+] Lack of address bus voltage convertersion

aleksander-martin, 2023-02-12:
According to Teensy 4.1 specs all pins are not 5V tolerant. I have noticed that only data bus is buffered between 5V and 3.3V with two 74LVC573 latches. Address bus is connected directly which means 5V signals will reach Teensy inputs and can theoretically damage it. Moreover I have found that MCL64 version intended for Commodore C64 has buffers for both data and address bus. Could you please explain this?

MicroCoreLabs, 2023-02-12:
The 6502 address bus is an output, so if there is no pull-up to 5V or another driver on the bus, a buffer is not needed. The Teensy's 3.3V output level will be recognized as a '1' by the receiving chips.     The 6510 used on the Commodore 64 has address lines which need to go hi-Z and have multiple 5V drivers on the bus, so a buffer is needed.

aleksander-martin, 2023-02-12:
OK :) Thank you for the answer!

closed 2023-02-12 by aleksander-martin