MicroCoreLabs/Projects #16: Chip orientation?  Black Screen.

COREi64, 2023-04-28:
I've soldered up one of these MCL64 boards, and am getting a black screen on power up on a 64.  I programmed the Teensy successfully, and am certain there are no bridges (did the soldering under high magnification under a microscope).  Triple checked my soldering, and used the mouser part numbers from the BOM.

The only thing I can think of is that somehow I haven't oriented the chips correctly as I don't see anything on the board that indicates where pin 1 should be.  I assumed that the top of the chip, with the pin 1 dot, is towards the Ux designation for each chip.

If my chip orientation isn't an issue, I am otherwise at a loss as to why it doesn't work for me.  Any suggestions on troubleshooting would be appreciated.

MicroCoreLabs, 2023-04-29:
Are you running at 800 mhz and fastest optimization? You could add some Serial.print statements to see if it gets out of reset and fetches instructions.  Please check the chip orientation in the pcb files using KiCAD.

COREi64, 2023-04-29:
Thanks for your comment.

I have confirmed that the chips are installed in the correct orientation.  So far so good.

This was the first time I'd ever programmed a Teensy, so eventually I figured out what you meant (I think) by 800mhz and fastest optimization.

I have been programming the board with the Arduino application.  The settings for the Teensy 4.1 I used in this last test is:

Optimize: Fastest
CPU Speed: "816 MHz (overclocked)"

There doesn't appear to be a setting for 800mhz however.

This time, programming it with these settings, I got a change in behaviour (however still not working).

I used the MCL64_Fastest.c sourcecode.

I am seeing a screen now, however full of garbage and not responding to keyboard commands:

https://postimg.cc/gallery/4s7gf5b

Some of the garbled characters are flickering, so it appears there is perhaps some life in there... just not behaving correctly.  Any thoughts?

Not familiar with serial print statements... or using this teensy in this manner but willing to try.

COREi64, 2023-04-29:
I must add that the comment about not responding to keyboard commands is actually not true.  I was poking around on the keyboard, and I do notice that the garbled shapes to change when I press keys.  If I do a runstop/restore, the whole screen does briefly flinch/blink.  Pushing various keys does change the shapes on the screen... so perhaps something is running.

MicroCoreLabs, 2023-04-30:
I suggest using the cycle accurate version of the code with the mode variable set to 0 or 1.

MicroCoreLabs, 2023-04-30:
The fastest code is used for massive acceleration demos. The cycle accurate version is the one you want for now.

COREi64, 2023-04-30:
Is the variable you're talking about in the code, this line?

uint8_t   mode=0;

COREi64, 2023-04-30:
So, given my lack of success with the software, I decided to start doing continuity checks on the board.  Low and behold, I found a manufacturing flaw in the PCB (thanks JLCPCB).  There was a hairline copper connection between pad 16 and 17 on chip U2.  And I had to desolder U2 to find it too.  Even under the highest magnification on my Amscope, I was unable to see it.  I scratched between the two pads and sure enough, I saw copper with the first scrape.  SHEZ.

After ensuring there's no bridge anymore, I soldered the chip back on and it started working.

Now, I'm wondering if perhaps that short maybe damaged something, as I'm finding the cycle accurate isn't completely compatible.  Doesn't work with the RAD Expansion cartridge or the Sidekick64 cartridge.  Doesn't work with the deadtest cartridge.  Found it does work with a few games I've tried loading with a 1541).

The Maximum Acceleration in turn isn't functioning well for me at all.  Even doing a small program to add 1 to a number, print to the screen and loop only runs for a short while, then give me an overflow error (it didn't even reach 500 when that happened).

I'm planning on building another just in case that bridge damaged something.  The Teensy still programs ok, so I have my fingers crossed that the short didn't damage that microcontroller.

MicroCoreLabs, 2023-04-30:
Im glad to hear you got it running.  With cycle accurate mode=0 I can use EasyFlash and an external 1541 drive successfully. I have run DeadTest from an image loaded in the Teensy, but I dont own an external cartridge.  Good idea trying another board.  A few other folks had luck with the MCL64 and cartridges, and mine worked on multiple C64's.

Thx,
Ted

COREi64, 2023-04-30:
I was so thrilled to see the machine spring to life!  I do appreciate your assistance and for sharing this awesome project.  Next week, I'll build another board, and use a new Teensy.  I'd be doing my continuity checks FIRST this time! =)  

I've had a few people ask me if I would build them one of these boards, however, wanted to check with you first if you would have any objection?  Also wanted to build one successfully first too!  Thought it might be nice to offer assembled boards on my shop (I run corei64.com).  Wondering if you would be amenable to me doing that?

I think it's such a cool board, but many people cannot solder things this fine.

My next build is going to be the MCL65+  I want to see how it performs in my VIC-20 and PET!

MicroCoreLabs, 2023-04-30:
Prolly should contact me directly through my website.

Thx
-Ted