MicroCoreLabs/Projects #20: Internal_RAM fixed to 256K for MCL86+?

HeathenUK, 2023-06-26:
Hi Ted,

Great work on all of these projects.

With the Teensy having 1024KB of RAM (albeit only 512KB tightly coupled) and with relatively mild RAM usage in the main accelerator, is there any reason we couldn't have Internal_RAM be 0x80000 or even 0xA0000? The latter would bust the tightly coupled RAM but should still fit?

Regards,

Greg

MicroCoreLabs, 2023-06-26:
Definitely could mirror address ranges in the Teensy 4.1 - I have done so.  Just need to be cautious of ranges which DOS DMA's data to and from. When this happens the RAM you are mirroring is no longer coherent with motherboard RAM.

HeathenUK, 2023-06-26:
I wonder if there's any way to track DMA calls. Of course part of the point is that the 8088 is out of the loop...

In any case, I don't currently have any inbound DMA to worry about!

G

MicroCoreLabs, 2023-06-26:
*   I wonder if there's any way to track DMA calls. Of course part of the point is that the 8088 is out of the loop...

Well, the MCL86+ can see every bus transaction so wouldnt be hard to sense when a DMA is being queued. Then after the DMA completes it could perform its own DMA which would copy the modified DRAM into the mirrored RAM in the Teensy.

HeathenUK, 2023-06-26:
I suppose without access to the non-8088 bus signals you'd need to trap writes to the I/O ports for DMA control registers and channels 1 and 3 (and maybe 2, for compatibility with floppy drives?)

This is just me being slow but I hadn't actually realised how simple the 8237 DMA controller actually is in that regard: https://www.lo-tech.co.uk/wiki/8237_DMA_Controller

G

HeathenUK, 2023-06-29:
Related - I've just clocked that the various bits of runtime (mode==1) conditional code in the accelerated version are commented out. It looks like most of that is about ROM and RAM shadowing though?

I gather from this that the intended function now is to hard set mode = 1 at compile time rather than via serial at runtime? That makes sense especially re. ROM mirroring.

The only piece of code I'm less clear on which is now commented out is `//if ( mode==1) clock_counter=0;`, which I think is intended to drain the pseudo op cycle counter to maximise the acceleration?