MicroCoreLabs/Projects #23: Wheelwriter project, errors synthesizing

johnwiggins, 2023-09-10:
I get an error trying to synthesize mclwr1.v in Diamond 3.6 or 3.12:

documents/impl1/source/mclwr1.v(120): net rx_fifo_wr is constantly driven from multiple places at instance RX_FIFO, on port WrEn. VDB-1000
Done: error code 2

And programming directly from the JEDEC in the repo produces transmit checksum errors and results in a non-functional board.

MicroCoreLabs, 2023-09-10:
Did you create FIFO cores using the Lattice IP tool?

johnwiggins, 2023-09-10:
Ouch, I did not. Sorry, newb here.

In the IPexpress tool, it looks like FIFO_DC is available for the XO2, but not FIFO. Is that sufficient?

Creating FIFO_DC (1024 depth) and including them in the synthesis cleared the error.