MicroCoreLabs/Projects #25: EPROM Emulator max size?

muramasatheninja, 2024-04-25:
Hi, I just stumbled upon your eprom emulator project. It looks like at the moment the max size is 64KB (27C512). I have a project that needs 512KB (27C040). I wanted to see if it would be possible to scale this design up for that and if you had any recommendations to accomplish this?

GrantMTG, 2024-04-26:
I don't know where you are going to get the pins from. Maybe a larger form factor Teensy, if there is such a thing. Check out the schematic:

https://github.com/MicroCoreLabs/Projects/blob/master/EPROM_Emulator/ROM_Emulator_PCB_CE_n/ROM_Emulator.pdf

MicroCoreLabs, 2024-04-27:
If you adapt the design to the Teensy 4.1 you should have enough pins.  Will be a larger PCB however...