MicroCoreLabs/Projects #28: MCLZ8: four-layer pcb necessary?

r00t-, 2024-12-04:
i was going to have the MCLZ8 PCB fabbed,
but then noticed the gerbers you uploaded are routed with four layers
(which makes the board quite a lot more expensive),
while thare are actually only three traces on the middle layers:
[image]
i'm probably going to re-route the board with two layers,
but was wondering if you had a specific reason to use four.
would you say the ground plane is required for the circuit to work?

AWCharlie, 2024-12-04:
JLCPCB has almost no difference between 2 and 4 layers. Click the 1-4
layers box. 5pcs for a couple bucks  https://jlcpcb.com/

r00t-, 2025-01-08:
thanks for your response(s).
it turned out that jlcpcb displayed an incorrect (absurdly expensive) quote due to some glitch on their website.
i now received ten boards (four layers) for a bit over 10eur (with some black friday special offer).
(if anybody in germany/europe needs one, i can share.)
now waiting for the other parts to assemble the board!

[image]

closed 2025-01-08 by r00t-